QuickSander/ArduinoHttpServer #9: StreamHttpRequest.hpp Syntax Error

vicyap, 2017-10-12:
```
In file included from lib/ArduinoHttpServer-0.1.1/src/ArduinoHttpServer.h:11,
                 from .build/uno/src/blinky.cpp:2:
lib/ArduinoHttpServer-0.1.1/src/internals/StreamHttpRequest.hpp: In constructor 'ArduinoHttpServer::StreamHttpRequest<MAX_BODY_LENGTH>::StreamHttpRequest(Stream&)':
lib/ArduinoHttpServer-0.1.1/src/internals/StreamHttpRequest.hpp:110: error: expected `(' before '{' token
lib/ArduinoHttpServer-0.1.1/src/internals/StreamHttpRequest.hpp:110: error: expected `;' before '}' token
lib/ArduinoHttpServer-0.1.1/src/internals/StreamHttpRequest.hpp: At global scope:
lib/ArduinoHttpServer-0.1.1/src/internals/StreamHttpRequest.hpp:110: error: expected unqualified-id before ',' token
lib/ArduinoHttpServer-0.1.1/src/internals/StreamHttpRequest.hpp:111: error: expected constructor, destructor, or type conversion before '(' token
```

QuickSander, 2017-10-13:
Will look into this. Seems like the compiler you use cannot handle the POD initialize syntax with curly brackets.

QuickSander, 2017-10-13:
What environment do you work with? So I can checkout the compiler it uses. Probably not a C++11 compiler.

vicyap, 2017-10-13:
Yes, it seems your suspicion is correct.

I am using `macOS 10.12.6` and the `Arduino Classic IDE 1.0.6`. That IDE ships with `avr-gcc-4.3.2`.

It works after I upgraded my compiler to `avr-gcc-4.9.2`.

closed 2017-10-13 by vicyap

QuickSander, 2017-10-13:
Good to hear. Thanks.