QuickSander/ArduinoHttpServer #15: 'Method' is not a member of 'ArduinoHttpServer'

thalesfsp, 2019-07-11:
Tried to run the example, and got this compilation error.

QuickSander, 2019-07-12:
Please paste the compilation error. Here it compiles fine. In fact, on Travis it is compiled on every pull request. 

It is highly dependent on the kind of environment you use. With platform.io it should work out of the box.

thalesfsp, 2019-07-12:
Hi! I'm using the latest Arduino IDE, not Platform.io IDE and trying to run the example in the Arduino MKR1010 WiFi, latest board firmware. About the error, this is the error :)
`'Method' is not a member of 'ArduinoHttpServer' `

QuickSander, 2019-07-15:
Hi,

I have to make a guess here so sorry for any mistakes, but are you talking about:

```
enum class Method : char
{
   Invalid, Get, Put, Post, Head, Delete
};

```
Then you need to refer to it like the following:

`ArduinoHttpServer::Method method( ArduinoHttpServer::Method::Invalid );
`
You need to make sure you have a C++11 compatible compiler.

I hope this helps.

closed 2019-07-16 by thalesfsp