QuickSander/ArduinoHttpServer #18: Compile issue using Arduino IDE 1.8.13

theo-stidsen, 2021-04-18:
When I include the library in my code, I'm unable to compile...

`In file included from C:\Users\stidsen\Documents\Arduino\libraries\ArduinoHttpServer\src\internals\HttpResource.cpp:10:0:
C:\Users\stidsen\Documents\Arduino\libraries\ArduinoHttpServer\src\internals\HttpResource.hpp:23:24: error: 'String' does not name a type
     HttpResource(const String& resource);
                        ^~~~~~
`

I worked my way around the problem by:

1. Changing the order of `#include "FixString.hpp"` and `#include "Arduino.h"` in HttpField.hpp
2. Replaced `#include <WString.h>` by `#include <Arduino.h>` in HttpResource.hpp

QuickSander, 2021-04-18:
Interesting that it works without trouble in platformIO. The order of `Arduino.h` and `FixString.hpp` I did on purpose so I'm sure FixString does not secretly rely upon indirect induces from Arduino.h. So that should be the correct order.

It is maybe some post processing / special handling that the Arduino IDE does, but normal compilation using standard tools works fine.

QuickSander, 2021-04-18:
```
Compiling .pio/build/esp01/libb36/ArduinoHttpServer/internals/HttpField.cpp.o
Compiling .pio/build/esp01/libb36/ArduinoHttpServer/internals/HttpResource.cpp.o
Compiling .pio/build/esp01/libb36/ArduinoHttpServer/internals/HttpVersion.cpp.o

Linking .pio/build/uno/firmware.elf
Checking size .pio/build/uno/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  29.2% (used 597 bytes from 2048 bytes)
Flash: [=====     ]  52.0% (used 16788 bytes from 32256 bytes)
Building .pio/build/uno/firmware.hex
============================================================================= [SUCCESS] Took 7.76 seconds =============================================================================

```

theo-stidsen, 2021-04-18:
Tried to replace `Arduino.h` with `WString.h` (as suggested) which throws me the below error...

```In file included from C:\Users\stidsen\Documents\Arduino\libraries\ArduinoHttpServer\src\internals\HttpField.hpp:16:0,
                 from C:\Users\stidsen\Documents\Arduino\libraries\ArduinoHttpServer\src\internals\HttpField.cpp:10:
C:\Users\stidsen\Documents\Arduino\libraries\ArduinoHttpServer\src\internals\FixString.hpp:33:29: error: '__FlashStringHelper' does not name a type
    explicit FixString(const __FlashStringHelper * pFlashStr);
```

QuickSander, 2021-04-18:
Oké thanks for the attempt. I will need to install Arduino IDE to see what's happening.

adamwitiak, 2022-08-05:
I am having @theo-stidsen 's issue, but in PlatformIO. Here's the error message:

`In file included from .pio\libdeps\wifi_enabled\ArduinoHttpServer\src\internals\HttpField.hpp:13:0,
                 from .pio\libdeps\wifi_enabled\ArduinoHttpServer\src\internals\HttpField.cpp:10:
.pio\libdeps\wifi_enabled\ArduinoHttpServer\src\internals\FixString.hpp:33:29: error: '__FlashStringHelper' does not name a type
    explicit FixString(const __FlashStringHelper * pFlashStr);
                             ^~~~~~~~~~~~~~~~~~~
.pio\libdeps\wifi_enabled\ArduinoHttpServer\src\internals\FixString.hpp:34:29: error: 'String' does not name a type; did you mean 'FixString'?  
    explicit FixString(const String& arduinoStr);
...
*** [.pio\build\wifi_enabled\libb97\ArduinoHttpServer\internals\HttpField.cpp.o] Error 1
========================================================== [FAILED] Took 1.92 seconds ==========================================================`

QuickSander, 2022-08-05:
Could you post your platform.ini file? Then I can see which environment settings you have and then I should be able to mimic it (My build using the Arduino framework with a ESP8266 huzzah board compiles. Most probably the FlashStringHelper class doesn't exist for the board you use).

QuickSander, 2022-08-05:
I've expanded the option to disable flash usage to also disable the constructor it complains about. Could you update the library to the latest version and recompile? Probably the next thing complaining will be the ```String``` type which should come from ```WString.h```. Something seems out of the order with your configuration.