QuickSander/ArduinoHttpServer #24: I have ported ArduinoHttpServer to Arduino Uno R4 WiFi, there is not much work, just few little changes

openshwprojects, 2023-08-19:
Here is the detailed writeup:
https://www.elektroda.com/rtvforum/topic3996584.html
You can also see my pull request: 
https://github.com/QuickSander/ArduinoHttpServer/pull/23

QuickSander, 2023-08-19:
Thanks for the nice write up.

The Base64 lib is indeed an awkward not that often used version. You can also disable the complete BasicAuth code compilation by #define ARDUINO_HTTP_SERVER_NO_BASIC_AUTH.

This successfully removes the need for Base64 library dependencies as a whole.

Somewhere in the future I want to convert the used Base64 lib to a more commonly used one.

openshwprojects, 2023-08-19:
That's good to know. 

I can open a pull request with R4 compatibility, I just need to find out which #define  is used to mark R4 platform as current, so I can switch some things, like, for example, use WiFiS3.h instead of WiFi.h