QuickSander/ArduinoHttpServer #25: Base64 compile error with Arduino RP2040 Connect

leopold-gravier, 2023-09-02:
Hello,
Whenever I try to include ArduinoHttpServer on a project for an Arduino RP2040 Connect, it fails to compile.

Ex :
```Arduino
#include <ArduinoHttpServer.h>

void setup(){}
void loop() {}
```

Output :
```
In file included from C:\Users\L�opold\Documents\Arduino\libraries\ArduinoHttpServer\src/ArduinoHttpServer.h:10:0,
                 from C:\Users\Léopold\Documents\Associatif\Minotaure\Projet Initiation\Code\Test\Test.ino:1:
C:\Users\L�opold\Documents\Arduino\libraries\ArduinoHttpServer\src/internals/StreamHttpRequest.hpp: In member function 'bool ArduinoHttpServer::StreamHttpRequest<MAX_BODY_SIZE>::authenticate(const char*, const char*) const':
C:\Users\L�opold\Documents\Arduino\libraries\ArduinoHttpServer\src/internals/StreamHttpRequest.hpp:403:30: error: 'Base64' was not declared in this scope
    const int encodedLength = Base64.encodedLength(combinedInput.length());
                              ^~~~~~

exit status 1

Compilation error: exit status 1
```
[The error message with verbose enabled](https://github.com/QuickSander/ArduinoHttpServer/files/12504353/Error.message.with.verbose.txt)

I invastigated a bit and it seems very specific to the Arduino RP2040 Connect board, I tried compiling the same script for a regular Nano board and it worked fine.

leopold-gravier, 2023-09-02:
It appears that another "base64.h" file is loaded. There is one such file inside the libraries specific to the RP2040 Connect board.

The exact path is C:\Users/[Your username]\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\4.0.4\cores\arduino\mbed\connectivity\mbedtls\include\mbedtls\base64.h

leopold-gravier, 2023-09-02:
System info :
Windows 10
Arduino IDE 2.2.1
ArduinoHttpServer 0.10.1
Base64 1.0.0

sholsapp, 2024-01-20:
I have the same issue with a ESP8266 board, which also has its own base64.h implementation in the core.

leopold-gravier, 2024-01-22:
Thank you for your answer @QuickSander :)

Although, is this flag implemented in the latest release ?
I tried it with the latest release available on the arduino IDE (ArduinoHttpServer 0.10.0), but I had the same error when compiling.
```Arduino
#define ARDUINO_HTTP_SERVER_NO_BASIC_AUTH
#include <ArduinoHttpServer.h>

void setup(){}
void loop() {}
```

It is only when I installed the library manually using this repo that it worked.

QuickSander, 2024-01-24:
Hi, yes it is indeed available in the version online: https://github.com/QuickSander/ArduinoHttpServer/blob/8bdc3d5a18c6a310ce3c9af62cce38e260f7c056/src/internals/StreamHttpRequest.hpp#L20